davidgiven/cpm65 #20: 6502 Basic

SEGStriker, 2023-05-06:
I don't see any implementation of Basic or Pascal. Well, Pascal will be harder to find or write a new one.
But 6502 Basic is already here. Can you compile it with your ASM.COM? I tried, but the format is too different.
[BASIC.zip](https://github.com/davidgiven/cpm65/files/11412689/BASIC.zip)

davidgiven, 2023-05-06:
I've been thinking about porting Microsoft Basic, but it looks complex --- ASM.COM has some bugs and missing features I'd need to implement. The one you linked is rather simpler. Where did it come from? It's clearly a 6502 CP/M port, so I'm guessing either DOS65 or DOS/65 (these are different projects)?

I can't use it as it stands because it has no license, which is why I need to know...

SEGStriker, 2023-05-07:
I got it from here: https://github.com/Dietrich-L/CPM-65/tree/main/BASIC

But, as seen in the source code, it is much earlier, written by another author.
Dietrich has his own port of CP/M-65, initially to only one platform. Some time ago I gave him an idea to port it to Apple II. Meanwhile you also ported your version to Apple II. Actually, it works on Apple II also (not only //e, but prints only the odd symbols, others are sent to non-existing 80-column card. Not sure if it will work with Videx in slot 0.

davidgiven, 2023-08-15:
I completely forgot about your message! And then went and ported Altirra BASIC, which is an Atari BASIC clone and significantly more work than porting this would have been... but at least there's a BASIC now, even if it's not very Microsoft compatible.

davidgiven, 2023-08-16:
Unfortunately, looking at Dietrich's code in more detail, it's pretty clear that it's not original and is in fact reverse engineered from Microsoft BASIC. e.g.:

From the leaked Microsoft BASIC code:

```
FRE:
        lda     VALTYP
        beq     L3188
        jsr     FREFAC
L3188:
        jsr     GARBAG
        sec
        lda     FRETOP
        sbc     STREND
        tay
        lda     FRETOP+1
        sbc     STREND+1
```

From Dietrich's code:

```
FRE   LDA VAR_F      ;FRE COMMAND
   BEQ FRE1
   JSR GARCOL1
FRE1   JSR RM_STR
   SEC
   LDA STREND
   SBC VTEND
   TAY
   LDA STREND+1
   SBC VTEND+1
```

There are many other similarities. It's even got the "MICROSOFT!" Easter Egg in it. So that renders it unusable, I'm afraid.

closed 2023-08-16 by davidgiven

davidgiven, 2023-08-17:
I doubt it, unfortunately. They look far too similar to be a coincidence. Even the token names are listed in the same order.

SEGStriker, 2023-08-17:
So, I guess you aren't supposed to use the official MS code?