davidgiven/cpm65 #23: Device driver model

davidgiven, 2023-05-15:
Looks like we're going to want this soon, so I'm going to rough out the design here...

The rationale is that various applications need different functionality that we don't want in the base BIOS because it occupies RAM. Example: a terminal emulator. These aren't cheap, and we shouldn't require users to have it in memory if they're not using it. In addition, the original CP/M was hard to extend, leading to a miserable experience when trying to, e.g., change the serial port baud rate. Let's not make that mistake here. At the same time, let's try and keep things as lightweight as possible. Conversely, we also want to be able to inline drivers into the BIOS (e.g. for a screen driver).

So the basic concept is: the BIOS keeps a singly-linked list of drivers. Drivers can be looked up by ID, returning a management routine. An application which wants to use a driver will look it up, stash the  address, and can then call routines on it. An application which wants to _install_ a driver can register a new driver on the top of the list, then adjust the memory bounds before returning so the driver code stays resident.

Each driver has a management routine for doing stuff like device lookup and enumeration, and one or more strategy routines, which care entirely driver-specific.

New BIOS entrypoints:

 - BIOS_ADD_DRIVER: writes to the top-of-list address, returning the old address.
    Intention: to install a driver, you call this, providing the new driver's management routine. The old address gets stored for use by the management routine.
    This would also cause the BIOS to recache any strategy routines it may have (see below).

  - BIOS_FIND_DRIVER: takes a 16-bit driver ID. Looks up a driver and returns its strategy routine address, or errors out if there isn't one. (This just calls DRV_FIND on the top driver below.)

That's all there is.

The management routine takes an opcode in Y and a parameter in XA.

  - Y=0: DRV_FIND: given a driver ID in XA, returns its strategy routine address, or 0.
    This compares the name with the driver's own. If it matches, the driver's strategy routine is called. Otherwise, the routine delegates to the next one on the list via the address stored when BIOS_ADD_DRIVER was called.
    A driver may respond to multiple device IDs by returning different strategy routines. e.g. a serial driver can handle multiple devices.
  - Y=1: DRV_ENUMERATE: given a function pointer in XA, calls it once with each device ID supported; then it passes control onto the next driver on the list.
    (I would omit this to begin with. We may not need it at all. I can't really see much use for it other than diagnostics.)

The BIOS code therefore becomes almost trivial, with one pointer of storage (for the current head-of-list). DRV_FIND is a `cmp; bne; cpx; beq; jmp (...)`.

A driver example: the screen.

  - Y=0: get driver version number.
  - Y=1: get screen size.
  - Y=2: set cursor position.
  - Y=3: get cursor position.
  - Y=4: write string at cursor position.
  - Y=5: clear screen.
  - Y=6: clear vertical area (top and bottom lines are supplied).
  - Y=7: scroll up.
  - Y=8: scroll down.
  - Y=9: get key with timeout. (We'd need to define numbers for special keys like cursor keys.)
  - (optional, detected via version number): Y=10: set text style.
  
A TTY:

  - Y=0: get driver version.
  - Y=1: write character.
  - Y=2: read character with timeout.
 
(The BIOS would always supply a minimal TTY, which the BIOS console in/out routines would call. Loading a new TTY device would override the old one, and therefore the BIOS would start sending characters there instead. This new one could have a much more sophisticated terminal emulator state machine, implementing something like VT52, implemented by calling the screen driver; so making it platform independent.)

davidgiven, 2023-05-17:
I would prefer to omit these ones:

- delete character right from current position, and move remaining characters
to the left
- insert empty line, moves current line and all lines below one line down
- delete line, moves all lines below one line up
- get character at current position?

The rationale is that if the hardware doesn't support it, these can't be emulated without keeping a shadow copy of the screen (e.g. to insert a character you have to be able to redraw the entire line). This makes it hard to come up with a fallback solution --- consider, say, trying to write a screen driver on top of VT52...

Another command which would probable to useful is clear to end of line. Most terminals support this and I've used it myself for my CP/M-80 vi clone: https://github.com/davidgiven/cpmish/blob/master/cpmtools/qe.c It's also a useful primitive for implementing the other clear functions.

ivop, 2023-05-17:
I agree. What I wrote about 15. get character does indeed apply to 8, 9 and 10, too. Somehow it didn't occur to me yesterday. So omitted they will be.

ivop, 2023-05-17:
Hmm, I think insert and delete line should not be omitted. There are no special requirements for the backend. Just copying screen memory up or down, without knowing its contents as characters. It could be a bitmap. We need some form of scrolling besides line feed at the end of the page.

Perhaps I'm misunderstanding something. I was thinking the idea was to make this work:

Boot with default TTY.
Load platform specific SCREEN driver from the command line
Load ADM3A or VT52 TTY driver from the command line
Run program that needs an enhanced TTY.

davidgiven, 2023-05-17:
I'm thinking of devices which might not have video memory access. For example, an actual terminal! e.g. something like the Ben Eater 6502 SBC hooked up to a VT52 emulator.

I'm thinking that for a computer with video hardware you 'd have:

    BIOS (w/ basic TTY) -> SCREEN on real hardware -> full TTY

(although it's probably sensible to embed SCREEN into the BIOS as the basic TTY will likely share code). But for an SBC you'd have:

    BIOS (w/ full TTY connected to serial port) -> SCREEN emulator.

I suppose you could put another TTY on it if you wanted...

ivop, 2023-05-17:
Ah, I thought the idea of the screen driver was to abstract away the hardware details of the host system so vt52 can be platform independent. After boot, TTY talks to the hardware or host OS directly. After loading SCREEN plus VT52DRV, TTY talks to SCREEN which talks to the hardware or host OS.

Your SBC example with a full blown VT52 terminal connected to the serial port wouldn't need any of them. Just a raw serial I/O port in the BIOS to the actual terminal that handles all screen related stuff itself.

davidgiven, 2023-05-18:
Yes --- there's two scenarios here:

- machine with video hardware; it has a platform-specific SCREEN driver, which SCREEN-aware programs can use directly, and programs which want to speak VT52 install VT52DRV.
- machine with a serial terminal; it has a TTY driver, which programs which want to speak VT52 can use directly, and programs which want to speak SCREEN will need to install a generic SCREEN-on-top-of-VT52 driver.

...not that I really expect to see that last, but I would at least like to support it.

I've done a really basic SCREEN for the BBC Micro (because it already has a decent terminal emulator built in) in #27...

davidgiven, 2023-12-27:
Closing for cleanup.

closed 2023-12-27 by davidgiven