davidgiven/cpm65 #29: Problem getting C64 working on emulation

ivop, 2023-05-19:
Hi,

Yesterday I figured out how to get the VIC-20 and PET versions working on emulation with Vice (I already had b-em and linapple running, and of course atari800 and Altirra with wine), but I'm having problems with the C64. I got my ROMs here:

http://www.zimmers.net/anonftp/pub/cbm/firmware/computers/

I tried about every combination of kernal and 1541 ROMs I could think of, but none work. It seems my true drive emulation of the 1541 works correctly because it works with xvic. Any idea what might be wrong?

Yesterday it got stuck after printing one dot. Today it doesn't even print a dot and goes back to the ready prompt.

kernal.901227-03.bin
roms/characters.901225-01.bin
basic.901226-01.bin

1541-c000.325302-01.bin
1541-e000.901229-05.bin

True drive emulation enables
RPM: 300
Wobble: 3000
Amplitude: 2000

I tried disabling wobble completely, but that makes no difference.

Regards,
Ivo

davidgiven, 2023-05-19:
Here are my settings --- I've tried this with both a 1540 and a 1541 and they both work:

[image]

Silly question --- did you turn all the RAM blocks on in Machine->Model?

ivop, 2023-05-19:
For the VIC-20 I did enable all the extra RAM, and that worked correctly.

For the C64, I don't see anything RAM related under Machine->Model. They all have 64kB right? Under memory expansion hacks I can enable 256K but that doesn't make a difference.

It seems my version of Vice is newer as I have more options under Drive, namely memory expansions for the 1541. Enabling or disabling doesn't change anything though. I assume PAL C64 should be the Model?

davidgiven, 2023-05-19:
Sorry, I misread your message! I didn't realise you were referring to the C64. And yes, I've just tried it and it's broken. It goes straight back to the READY prompt. I'll take a look.

davidgiven, 2023-05-19:
From a list of 6502 bugs:

> An indirect JMP (xxFF) will fail because the MSB will be fetched from address xx00 instead of page xx+1.

Guess what's happening!

ivop, 2023-05-19:
I haven't stumbled upon that bug in thirty years. Good find!

While I'm debugging my SCREEN implementation, I set my screen width to 38. If I start qe bedit.txt and press j a few times, it hangs. Confirmed on the BBC with getsize forced to lda #38 rts.

BTW I'm considering doing a complete rewrite of the Atari TTY's conout and go directly to the screen memory. Its builtin E: (Editor) handler is way too slow. I mean 600 baud slow.

davidgiven, 2023-05-19:
I did actually find an unrelated bug caused by the TTY change --- the TTY jump table code and the BDOS loader both wanted to use the same pointer. Oops. Should be fixed now.

Re SCREEN: I screwed up the spec and will have to rework it. Sorry. (Basically, defining cursor-movement behaviour when printing characters makes things way too difficult to implement, especially with stuff like implicit wrap and scrolling, so I'm going to take all that out. I think it's easiest for the cursor not to move after any command, even putchar.)

ivop, 2023-05-19:
Okay, I'll stop working on my SCREEN implementation then. Stupid Atari E: handler scrolls all lines below one down when I print a character at the last column. That's why I was testing a screen less wide with qe. Another stupidity of Atari is that it doesn't move the cursor "glyph" if you call XIO Position. It moves it when you print the first character at the new position.

I'll see if I can do 'more' tonight. And I'd like something like copyuser filename usernumber:. We don't have PIP, and I don't think we want that ;) I think in the future it might be nice to have all disks for all platforms have a few user areas. One startup, one fonts+setfnt, one documentation+viewer, one assembly+editor+asm, etc... Directory listing is getting longer and longer. But then it'll be nice if you can copy your freshly assembled binary to user 0.

Or perhaps I'll rewrite conout first. I want it faster.

ivop, 2023-05-19:
BTW x64 c64.d64 now runs. Thanks for fixing it! Might also take a look at how redefining C64 fonts work. Can the BBC redefine its font? "vi" with divide symbol instead of ~ and pound sign instead of ` looks kinda funny.

davidgiven, 2023-05-19:
If you're seeing a `÷` instead of `~`, you're in mode 7 --- that's the 40x25 Teletext mode with an unmodifiable font. There's an 80x25 bitmap mode which works better and in which £ is redefined to backtick. 

If you're on a BBC B or a BBC B with Tube, do this:

```
MODE 3
*!BOOT
```

If you're on a BBC Master or a BBC Master Turbo, do this:

```
*CONFIGURE MODE 131
```

That'll set the boot mode in CMOS RAM so it'll always start up in that mode, so from then on SHIFT+BREAK works as usual.

Note that on a BBC B, mode 3 eats 16kB of RAM, so you might want to stick with the default mode 7. On a Tube or Master system, screen modes have their own bespoke video memory.

(I will copy this into the README...)

ivop, 2023-05-20:
Funny things happening again :)

BBC B (German or US)

Boot from the command line results in the CPM/65 banner being printed and then <space>isk fault.

BBC B+ 64K

Boot from the command line boots correctly. If I let it go to BASIC first and type *!BOOT, it hangs after the CP/M 65 banner.

With Tube enabled both are similar to BBC B+64K. Autoboot from the b-em Disc menu also fails. But autoboot from the Linux command line works.

davidgiven, 2023-05-20:
Yup, that was broken. Fixed in #32. I really need a way to automate tests. I suspect this bug was the cause of all the weirdness you've been seeing.

MODE 4 is 40x32 and uses 10kB. You might prefer MODE 6, which is 40x25 and uses 8kB.

davidgiven, 2023-05-20:
I've pushed the SCREEN changes --- PTAL? I think the current feature set is a reasonable Minimum Viable Product for such a thing. There's what's best described as 'documentation' in driver.inc. Plus, there's a PET driver which draws to screen memory directly which you might find useful.

ivop, 2023-05-21:
Nice work! I didn't have the time to look at it (got sidetracked by another project) but I update my llvm-mos-sdk from your fork and got it running. All BBC models now run correctly and also boot from the basic prompt. I'll have a proper look later today and see if can start the Atari SCREEN implementation. Hopefully I can indeed copy large parts from your PET port ;)

davidgiven, 2023-12-27:
Belatedly --- good to know, closing for cleanup.

closed 2023-12-27 by davidgiven