davidgiven/cpm65 #37: Generic load BDOS from CP/M filesystem

ivop, 2023-05-22:
Hi,

You mentioned a generic BDOS loader in my last pull request comment section. This would indeed be really helpful in case the size of BDOS changes and avoid having to adjust the bootloader and disk image generation, at least for the Atari. Depending on the amount of assumptions we allow, this could be done in a few lines of code.

* assume BDOS.SYS is first file written to the CP/M fs
* it is always the first directory entry
* it is stored in consecutive sectors after the directory
* we need to adjust for reserved sectors (hardcoded, derived from dph)
* retrieve first directory sector
* extract ex counter
* load that amount of sectors into  __USERTPA_START__
* relocate
* run

Were you planning on doing this yourself? If not, I could take a stab at it.

Regards,
Ivo

ivop, 2023-05-22:
Proof of concept:

https://github.com/ivop/cpm65/tree/bdos-loader

ivop, 2023-05-22:
Proof of concept. Created a pull request here: #39

davidgiven, 2023-05-22:
Ah. I just did it in #40!

It'll actually load any one-extent file with arbitrary block numbers (so the file can be fragmented). There are currently two versions, for large disks and small disks, but I'm not sure it's worth it --- it ended up being only 245 (discardable) bytes. I was able to drop one of the reserved tracks on the Apple II with this, which is great because it opens the door for data disks which don't have the BDOS and CCP on them (although they'll still need the reserved track).

ivop, 2023-05-22:
Sorry, I couldn't resist trying it out myself ;)

My loader is 71 bytes, but it has more restrictions.

davidgiven, 2023-05-22:
The intention is that the loader gets discarded after use, so it doesn't matter how big it is. It looks like the Atari port doesn't discard anything, so it's not appropriate there, so I have no objection to using yours.

(Although I think the Atari port could be made to discard its init code --- in the boot sector you get to specify the load address, right? So you could do the intial load somewhere high, then copy the resident bits down to their final address, before loading the BDOS. When you run it, all the init code gets overwritten. The PET ports do this.)

ivop, 2023-05-23:
I have converted it to your new loader anyway. I like it to be consistent with the other ports. Right now I only discard part of the init code by overwriting it with the directory buffer. I'll look into loading it somewhere around $3000 and copy just the bios down to $0500. Should still work on a 16kB 400, 800 or 600XL then.

davidgiven, 2023-06-02:
(closing for cleanup)

closed 2023-06-02 by davidgiven