davidgiven/cpm65 #43: cpm_ram and gettpa behaviour

ivop, 2023-05-23:
Hi,

I stumbled upon some strange behaviour which I cannot explain.

See this code in my test branch: https://github.com/ivop/cpm65/blob/test/apps/test.c

Output:

```
    CP/M-65 for the Atari 800                   
                                                
    A>free                                      
    ZP: 95 to FF. Free: 6A                      
    TPA: 1900 to BC00. Free: A300               
    CCP at: B200                                
    A>test                                      
    start: 9500                                 
    cpm▁ram: 3790                               
    end: ff00                                   
    A>                                          
```

```
$ xxd -s 1 -l 1 .obj/test.com
00000001: 21                                       !
```

I expect the new tpa_start to be $1900 + $2100  = $3a00

Any idea what's going on?

davidgiven, 2023-05-23:
The most likely explanation is that the C bindings are using the wrong BDOS entrypoint numbers and have ZP and TPA swapped. It would actually be easier to change the BDOS itself...

ivop, 2023-05-23:
You are correct.

calling cpm_bios_gettpa() results in:

```
    CP/M-65 for the Atari 800                   
                                                
    A>test                                      
    start: 1900                                 
    cpm▁ram: 3790                               
    end: bc00                                   
    A>                                          
```

Still, I expected the start address being updated by CCP before calling the loaded application, so you can easily chain load overlays if you want to. Is this intended behaviour?

davidgiven, 2023-05-24:
Fixed.

Re the CCP not updating the start/end address: that's intentional, as it makes it easier to know when the application has changed one of the addresses if it, e.g., wants to install a TSR. Adjusting it doesn't help much for overlays as only the application knows how much heap memory it needs.

There really needs to be some proper documentation...

PR #45 (2023-05-24): Fix incorrect BDOS_GETTPA and BDOS_GETZP calls.
These were backwards.

Fixes: #43
closed 2023-05-24 by davidgiven