davidgiven/cpm65 #46: BDOS and CCP outside TPA area

ivop, 2023-05-24:
Hi,

The Atari 800XL and later had 64kB of RAM instead of the 48kB maximum of the 400/800 series, but 16kB was "under" the ROM. The ROM can be switched off to access it, but there is a hole were the hardware registers live.

Current CP/M memory map:

0x0000 - 0x04ff zp, stack, Atari OS variables
0x0500 - 0xbbff BIOS, BDOS, TPA, CCP
0xbc00 - 0xbfff DL, screen memory
0xc000 - 0xcfff ROM
0xd000 - 0xd7ff hardware registers
0xd800 - 0xffff ROM

For 800XL and beyond I'd like to lay it out like this:

0x0000 - 0x04ff same
0x0500 - 0xcfff BIOS, DL+screenmem, TPA (screen memory cannot be under ROM, 4kB extra TPA)
0xd000 - 0xd7ff hardware
0xd800 - 0xffff BDOS, CCP

What would be an acceptable way to accomplish this? BDOS does not need to claim TPA and warmboot never needs to reload CCP.

After that, I'd also like to implement a 130XE (or expanded XL) version with an 80-column screen in an extended memory bank that's only visible to ANTIC (the screen processor). Needs ~8Kb or RAM for 320x192 graphics mode, but won't cost you any TPA space if you use extended memory. CPU/ANTIC access can be told to look at base RAM or extended RAM independently.

Regards,
Ivo

davidgiven, 2023-05-24:
This is the exact use case for SETBANK (the VIC-20 and Apple II use it, as will the Oric, eventually). There are two banks: MAIN and EXTRA. User applications go into MAIN, and the BDOS gets loaded into EXTRA. Usually both banks allocate out of the same pool, so the BIOS can just ignore calls to SETBANK. But on these systems, you can set EXTRA to point at the RAM from 0xd800 up. Eventually there'll be a LH CCP command for loading user programs into EXTRA, if available, so TSR device drivers can go into EXTRA too.

You can just cut-and-paste the VIC-20 code for the BIOS here; it's trivial. The hardest part is getting the BIOS into the extra RAM, if you want it to, as this can require some creative linker scripting. The Apple II has this vile two-stage link so that the BIOS can get loaded at the top of upper memory, letting it claim the vectors --- the linker doesn't support allocating from the top down, unfortunately.

Right now the CCP always loads into MAIN because it gets reloaded on every warm boot, but there's no reason why it can't be persistent. There just needs to be a way to tell the BDOS not to reload it.

I have, BTW, just pushed a change which enhances the CCP's FREE command to show both banks. Also I might change the bank names because right now they're kind of stupid.

ivop, 2023-05-24:
Thanks! I'll look into. Getting more familiar with CP/M by the day :) It's a pitty both the C64 and Atari chose to put their hardware registers at 0xd000, possibly just to save a few transistors. 0xf000 would have been much nicer.

Re putting BIOS there, I can probably get away without claiming the vectors. On the Atari there has to be a small resident BIOS portion in lowmem TPA for keyboard input and disk I/O. They need the Atari OS ROM enabled and IRQs for serial I/O, and you get PBI hardware extensions for free, like CF and SD cards and high speed SIO. The rest can be "under" the ROM.

davidgiven, 2023-12-27:
Closing for cleanup.

closed 2023-12-27 by davidgiven