davidgiven/cpm65 #53: Release images

ivop, 2023-06-16:
Hi,

Could you also add atari800xlhd.atr to the released disk images? It is currently missing. It's the XL/XE specific port I did recently.

Working on porting the 8080 emulator to CP/M-65, but the warm weather is slowing me down. Also had to do a few tunes for an upcoming game, and did some tests for an 80 columns mode utilizing 3x6 characters and a modified display list to reduce its memory footprint from 7680 to 5760 bytes (excluding the font and driver itself). More on that later.

Regards,
Ivo

davidgiven, 2023-06-17:
That's done.

3x6 characters? Yikes! I thought the 4x8 font used for the VIC-20 port was tight. The only 3x6 fonts I've seen were actually 4x6 with a blank column for readability... also, you've probably seen it, but if you haven't: the tools directory contains a library for reading BDF font files. See fontconvert.c, which uses this to generate the VIC-20's 4x8 font bitmap.

closed 2023-06-17 by davidgiven

ivop, 2023-06-17:
3x6 is indeed 4x6 with a blank column, and becomes 4x8 with a blank line on top an at the bottom, done by the display list so it doesn't cost any screen memory.