davidgiven/cpm65 #54: Memory banking issue in Apple 2e

univta0001, 2023-06-21:
In the Apple 2e CPM-65, it is not working in Apple 22JSE (https://www.scullinsteel.com/apple/e) because the bank switch is not working.

The source code is using two STA $C08b to switch the bank. Based on "Understanding the Apple IIe, by James Fielding Sather, Pg 5-24", bank switch should only work with reading twice on $C08b. Manually changing editing the image with two LDA $C08B fixes the issue.

Excerpt from [apple2e.S](https://github.com/davidgiven/cpm65/blob/master/src/bios/apple2e.S)

sta MEM_BANKING+0xb     ; R/W 0xe000 RAM; bank 1 in 0xd000 <- should be changed to lda MEM_BANKING+0xb
sta MEM_BANKING+0xb     ; yes, I'm sure <- should be changed to lda MEM_BANKING+0xb

univta0001, 2023-06-21:
As far as i know it should not work in the real hardware as bank switching will not be enabled with the STA commands. Using STA command, the wrt count will not reached 2 to cause the High RAM to be set.

[image]

davidgiven, 2023-08-15:
Sorry for the long delay --- but it seems this got fixed in April: https://github.com/davidgiven/cpm65/blob/master/src/bios/apple2e.S#L68 I think you must have found an old version from somewhere... so, closing.

davidgiven, 2023-08-15:
No it isn't. Misread your message, sorry!

commit 36739b84 (2023-08-15): Fix memory banking switch, which fails on real hardware.

Fixes: #54
closed 2023-08-15 by davidgiven