davidgiven/cpm65 #56: request: port this to the pC6502

snakebyte69, 2023-07-28:
I will send you one

quincy451, 2023-07-28:
It will be interesting if David Givens sees this board is a viable port.  
One thing I find interesting about this board is it has two RP2040 and the NEO6502 only has one.  Don't know what to make of that...might be a reason.

davidgiven, 2023-07-28:
Thank you very much! Where are you, though? I'm in Switzerland, so international postage would probably not make it worthwhile.

I've had a quick look at the docs and this seems reasonable --- the interface and memory map seem pretty well defined. It looks like storage happens via USB, using the Pico RIA as the host. Unfortunately documentation is very sparse and it's not clear how much of the API is supported, although I did find a file system implementation in the RIA source code.

For CP/M on this, the easiest thing would be to put a CP/M filesystem in a big file on the USB stick. It would be nice to have proper support for CP/M filesystems but I don't think there's any low level device API. Given that the FAT filesystem is free as it runs entirely on the RIA Pico, it's possibly not worth the effort.

snakebyte69, 2023-07-29:
My store ships world wide no problem