davidgiven/cpm65 #85: Build for Apple II 3.5" as well

SvOlli, 2023-12-18:
With the Apple II GS and the Apple II C plus, there are two machines capable of running CP/M 65, but can't due to a different disk format (3.5" instead of 5.25").

(And I'd really like to see CP/M 65 running on my real 2gs instead of just in an emulator.)

davidgiven, 2023-12-27:
That's the 800kB format, right? Making a disk image should be easy as IIRC it's just the same as the later Macintosh 800kB format, but they'll need a whole new set of disk access routines and I'm still burnt out from doing the Apple II routines! If you have any pointers to example code that'll help...

polluks, 2024-07-17:
SmartPort access it easy:
```
; Define SmartPort register addresses
SMARTPORT_COMMAND  = $C0E0
SMARTPORT_STATUS   = $C0E1
SMARTPORT_DATA     = $C0E2

; Command definitions
SMARTPORT_CMD_READ = $02
SMARTPORT_CMD_WRITE = $03

; Buffer for data transfer
BUFFER = $8000

; Initialize the SmartPort
INIT_SMARTPORT:
    LDA #$00
    STA SMARTPORT_COMMAND  ; Reset the SmartPort
    RTS
```
```
; Send a command to the SmartPort
SEND_COMMAND:
    LDX #$00
    STX SMARTPORT_STATUS   ; Clear status
    LDA #SMARTPORT_CMD_READ ; Load command (change to WRITE for write operation)
    STA SMARTPORT_COMMAND  ; Send command
    RTS
```
```
; Read data from the SmartPort
READ_DATA:
    LDY #$00
READ_LOOP:
    LDA SMARTPORT_STATUS
    BPL READ_LOOP         ; Wait until data is ready
    LDA SMARTPORT_DATA    ; Read data byte
    STA BUFFER, Y         ; Store in buffer
    INY
    CPY #$100             ; Assume 256 bytes to read
    BNE READ_LOOP
    RTS
```
```
; Write data to the SmartPort
WRITE_DATA:
    LDY #$00
WRITE_LOOP:
    LDA BUFFER, Y         ; Load data byte from buffer
    STA SMARTPORT_DATA    ; Write to SmartPort
    INY
    CPY #$100             ; Assume 256 bytes to write
    BNE WRITE_LOOP
    RTS

SEGStriker, 2024-08-20:
Any progress? I guess adding more drives support should be easy. For example S6 D2 (5.25"), S5 D1, D2 (3.5"), S7 D1, D2 (HDD or another disk type).

davidgiven, 2024-08-21:
I haven't actually touched it! Is 800kB disk access always through SmartPort or does it use a software-defined interface like the Apple II interface?

retrac0, 2025-06-16:
> I haven't actually touched it! Is 800kB disk access always through SmartPort 

For the Apple II/II+/IIe the controller has a coprocessor that does most of the work, and is accessed through Smartport.  On the IIc+ and IIGS the IWM controller is partially software-implemented -- but it's still wrapped with a Smartport interface in ROM.  If you wish to avoid all ROM routines, it would need the GCR translation tables etc. implemented.  The IWM does have a buffer to free it from the cycle-exact code needed with Disk II, but is otherwise much like the Disk II controller when directly programmed.  (It's also used in early Macintosh machines.)  There's documentation on it here: https://www.brutaldeluxe.fr/documentation/iwm/apple2_IWM_Spec_Rev19_1982.pdf

There are a few exotic third party controllers from the early Apple II days.  Nearly all mass storage (including 400K/800K and 1.4M floppies) on the Apple II is otherwise Smartport.

SEGStriker, 2025-06-17:
Well, there are 3.5" FDD controllers for Apple II series (and another one, integrated for Apple //c), also for clones, which I believe weren't Smartport. But as I never had such, I can't say from my own experience. Smartport controllers were later implementation, primarily for SCSI HDDs.