davidgiven/cpm65 #91: Saving files with BEDIT on the Atari doesn't work

davidgiven, 2023-12-20:
It seems the Atari always returns carry set, indicating that there's a disk error. Most other programs don't check for this, but BEDIT does. This makes the mametest script fail.

ivop, 2023-12-20:
I cannot reproduce this. What steps do you take?
I did:

bedit
10 lda #0x12
save test.asm
quit
dir test.asm

davidgiven, 2023-12-20:
I'm testing with MAME as part of the new emulator tests. Having done some poking I see that it fails differently on `a800` than it does on `a800xlp`. Try this:

```
mame a800xlp -flop1 atari800.atr  -console -rompath roms/ -autoboot_script src/arch/atari800/mame-test.lua
mame a800 -flop1 atari800.atr  -console -rompath roms/ -autoboot_script src/arch/atari800/mame-test.lua
```

(Running `sh scripts/get-roms.sh` will download all the required ROMs)

ivop, 2023-12-20:
Mame is notoriously bad for emulating the Atari 8-bit. Serial I/O is broken. Keyboard handling is broken. Both is done by Pokey and they have a half-baked Pokey emulation. Last time I checked (years ago) their pokey implementation was audio only, and even that was incomplete. Looks like not much has changed since. They intercept OS SIO calls but not mimic the OS correctly, like setting success flags. Similar problems with keyboard I/O. Try pressing backspace and see what I mean.

Retroarch has much better Atari 8-bit support, based on the atari800 emulator.