davidgiven/cpm65 #109: Reusing a static FCB for several file operations

venomix666, 2024-01-06:
I have an issue in vt52term, where I reuse a global static FCB struct for opening files for both sending and receiving with Xmodem. If I only open/close the files, I get strange behavior such as the file position from the previous file being remembered when opening a new file.

I currently have a working fix where I use memset to zero the entire FCB before starting a new file operation. This works fine, but is this the way it is intended to be done? I tried just setting ex to zero, but this did not solve the problem.

I guess this boils down to two questions:
1. Is it expected that the FCB must be cleared between a close and open operation, or is this a bug in the BDOS?
2. If it needs to be cleared, it a complete zeroing with memset needed or is there a smarter way to do this?

davidgiven, 2024-01-06:
You do need to clear some. I don't know why they're not cleared for you. See F_OPEN here: https://www.seasip.info/Cpm/bdos.html EX, S1, S2, RC and CR are the ones specified in the documentation. S2 is cleared for you in the code but I forget why.

For clearing the whole structure... memset and `fcb = {}` _should_ be equivalent in C. Really they should both produce a tiny embedded loop as that'd be more efficient than calling out to a function but I don't know whether llvm-mos does that or not.

venomix666, 2024-01-06:
Thanks for the clarification, then it is just my poor knowledge of CP/M which is the issue. I assumed that cpm_file_close would set up the FCB used by fcb_file_open or fcb_file_create for a new file.

The fcb = {} syntax is more elegant for sure. I am still getting used to these 'modern' (i.e. anything not supported by Watcom ;) ) additions to C.

venomix666, 2024-01-12:
Closed by #115

closed 2024-01-12 by venomix666