davidgiven/cpm65 #112: "REN"-like command for changing user ID

SvOlli, 2024-01-08:
How about adding a command like `REN` just for changing the user?

Something like `CHGU filename.ext 1` would change the user ID of the file in the current user ID to the specified one by changing the byte in the file system entry/entries. This would make the `USER` concept much more usable. If you want to have a file permanently on a different user ID, you can copy it first, `CHGU` it, and `REN` it on the new user ID.

I've taken a look at CCP.SYS and BDOS.SYS, but my ideas for a solution would involve a lot of copy/paste. Maybe you've got a better idea how to solve this?

Also, you might have got a better idea than `CHGU` to name the command.

Having 4MB of space in the filesystem on the Sorbus would really benefit from a less useless concept of user IDs.

ivop, 2024-01-08:
Hi,

Perhaps the copy utility could be modified. I don't know if this works or is even supposed to work, but I was thinking along the following lines:

1. open input file
2. save current user number
3. call change user
4. open output file
5. copy stuff
6. change user back to original user

davidgiven, 2024-01-08:
From the user perspective, the cleanest thing would probably to extend REN to take fully qualified names --- so `ren 1:newfile=0:oldfile`. Except, from the API perspective, CP/M's handling of user numbers is wretched and the underlying BDOS system call doesn't support this.

I did make some attempt to work around the user number stuff with the xfcb library but of course that only works with applications that use it (and it still doesn't support renaming). In general user names are horribly fragile because if you're working with multiple files you need to set the user number correctly before _every_ BDOS call or you can end up with extents on disk with the wrong user number...

The cleanest thing to do would be to change the FCB interface so that the user number was stored in it, but that's a pretty massive API change and is also incompatible with CP/M-80. So far I've been doing what most CP/M developers did and just try to pretend user numbers don't exist!