davidgiven/cpm65 #113: Multiple drive support

venomix666, 2024-01-08:
I have been experimenting with a BIOS for my homebrew computer with two drives (which in this case are two different cmpfs-files on a USB-drive), and it kind of works but I get som strange results:

- Changing drives with e.g. 'b:' does not work, I guess that this is not implemented in the CCP?
- Listing files on b: with dir only shows one file on the first attempt but then works fine apart from that the drive is shown as A: in the output.
- Reading and writing files on b: works fine, but CP/M crashes when it tries to warmboot after running an application from b:, or an application which accesses b: (e.g. 'copy asm.txt b:' crashes, but the copy operation is successful)

Is support for multiple drives fully implemented or am I trying to do something which is not supposed to work at this point?

davidgiven, 2024-01-08:
I have no devices which support this so it's never been tested.

The warmboot crash sounds suspiciously like it's trying to read the boot files from the wrong track. I bet there are several places which are failing to reselect the appropriate disk before doing an access.

venomix666, 2024-01-08:
Yes, this is likely what happens. I had a look at the BDOS code and it seems like it is assumed in entry_RESET that internal_LOGINDRIVE will select drive A: if the A register is 0, but select_active_drive is called without storing the A register to active_drive first.

I will try to store A to active_drive before the call to select_active_drive and see if that fixes the crash.

venomix666, 2024-02-16:
It works a lot better now - no more crashes at warmboot and changing drives at least changes the letter in the prompt. It still seems like the CCP always assumes that A: is the default drive though: 
* Changing to B: and running DIR still shows the content of A:
* DIR B: works in so far that it shows the contents of B:, but it still prints A: at the beginning of each line regardless of which drive is selected when the command is run.

I have been digging a bit in both the CCP and the BDOS without finding any obvious cause for this, but perhaps the FCB is set up with A as the drive when no drive is specified?

venomix666, 2024-02-17:
I looked a bit more now and I think that the current_drive variable in the BDOS is never set to the active drive number, it is just initialized to 0 and then never updated. 

This would cause bdos_GETDRIVE to always return 0 which would explain why A is always assumed to be the active drive.

venomix666, 2024-02-17:
The fix in the pull request I submitted solves the issue with A always being assumed as the active drive.

It does however introduce a behavior which I am not sure if it is expected or not - if an application accesses the second drive and does not change the active drive back before exiting, CCP will now be on the second drive. Example:
`A>dinfo B:`
`(...output from dinfo...)`
`B>`

Is this the expected behavior, i.e. is it up to the application to reset the active drive before exiting?

The issue with only one file being detected on the first access of the second disk is still present. It not only with DIR, the first access to the second drive after boot only finds the first file on on the disk.

davidgiven, 2024-02-17:
Thank you for the investigation! I've commented on the PR.

PR #200 (2024-10-30): Fix some BDOS and CCP issues.
- Ensure that the CCP and ls don't change drive when listing a directory.

This avoids issues where the current drive according to the BDOS doesn't match the current drive according to the CCP, which can make odd things happen.

Fixes: #151
Fixes: #113

- Fail with a hard error when trying to write to a full disk.

Previously it would just keep searching up off the end of the bitmap and trying to write to out-of-bound blocks.

Fixes: #170
closed 2024-10-30 by davidgiven