davidgiven/cpm65 #130: munmap_chunk(): invalid pointer

mrgw454, 2024-03-26:
Getting this error during the build process:

```
MKCBMFS src/arch/commodore+c64_cbmfs
munmap_chunk(): invalid pointer
make: *** [.obj/build.mk:1216: .obj/src/arch/commodore+c64_cbmfs/src/arch/commodore+c64_cbmfs.img] Error 1
make: *** [.obj/build.mk:1475: .obj/src/arch/commodore+pet4032_cbmfs/src/arch/commodore+pet4032_cbmfs.img] Error 1
make: *** [.obj/build.mk:1543: .obj/src/arch/commodore+pet8032_cbmfs/src/arch/commodore+pet8032_cbmfs.img] Error 1
make: *** [.obj/build.mk:1611: .obj/src/arch/commodore+pet8096_cbmfs/src/arch/commodore+pet8096_cbmfs.img] Error 1
make: *** [.obj/build.mk:1714: .obj/src/arch/commodore+vic20_cbmfs/src/arch/commodore+vic20_cbmfs.img] Error 1

```
Thank you for any help you can provide.

davidgiven, 2024-03-26:
Wut.

What system are you using?

mrgw454, 2024-03-26:
I am currently using a Debian 12 (Bookworm) system (arch64) with fresh copies of the following projects built from source:
```
cpmtools
llvm-mos-sdk
llvm-mos
vice (for cc1541)

```
... and from Debian repo:
```
fp
libfmt-dev
python3

```
I do build quite a few other retro based projects.  I thought a pointer error was strange as well.

davidgiven, 2024-03-26:
If you do `make V=1` it'll show you the commands lines it's invoking. Try running the mkcpmfs command with valgrind. With luck that will give more details.

closed 2024-07-14 by davidgiven