davidgiven/cpm65 #151: Multi drive support

ivop, 2024-08-25:
Hi,

I tried adding a second drive to the Atari port (https://github.com/ivop/cpm65/tree/atari-multi-drive) but it does not seem to work as expected:

[image]
[image]

Directory of A: works normally. I switch to B: and then the directory listing is not correct. CTRL-C and the DIR, still wrong. But DIR B: suddenly shows the correct directory. Any idea what's going on here?

Regards,
Ivo

davidgiven, 2024-08-25:
A lot of these issues got fixed in #113, but there's clearly something else wrong.

The symptoms look weird. If you look at the first sector of the directory of B:, are the entries for ATBASIC.COM and ATBASIC.TXT next to each other in the same sector? If so, then it looks like the last-visible-item variable in the BDOS is being somehow set to 1, which it shouldn't be. It's an optimisation to allow unused sectors of the directory to be skipped. It's supposed to be determined when the drive is logged on.

ivop, 2024-08-26:
More funny behaviour :)

[image]
[image]

I found out that when I type B: this code does not seem to work correctly:
```
    ; Decide if the drive was already logged in.

    lda login_vector+0
    ldx login_vector+1
    ldy active_drive
    jsr shiftr              ; flag at bottom of temp+0

    ror temp+0
    zif_cs
        jmp .    ; debug, endless loop
        rts
    zendif
```

The first time it should not take the zif_cs and continue after zendif because the drive was not logged in yet. But it thinks it does.

ivop, 2024-08-26:
Found it!

```
zproc internal_LOGINDRIVE
    ; Select the drive.

    sta active_drive          ; <----- MISSING
    jsr select_active_drive

    ; Decide if the drive was already logged in.
```

select_active_drive starts with lda active_drive, which is 0.

I have put sta active_drive within internal_LOGINDRIVE because elsewhere it is called with A set to what has to become the active drive.

[image]
[image]

venomix666, 2024-08-26:
Fantastic! This bug was driving me crazy when I tried to find the cause for it (see #113), I'm really happy to see it fixed.

ivop, 2024-08-26:
Thanks. There's still something fishy going on though.

[image]
[image]

Opening a file on a drive that has not been logged in yet does not work. I'll look into it later tonight.

ivop, 2024-08-26:
Think I found it. OPENFILE and CREATEFILE call convert_user_fcb. convert_user_fcb stores the drive number to active_drive and calls select_active_drive, which does not log in the drive. Changing that to internal_LOGINDRIVE (which calls select_active_drive, too) seems to fix this:

```
--- a/src/bdos/filesystem.S
+++ b/src/bdos/filesystem.S
@@ -243,13 +243,12 @@ zproc convert_user_fcb
     zif_mi
         ldx current_drive       ; override with current drive
     zendif
-    txa
-    sta active_drive            ; set the active drive
 
     lda current_user
     sta (param), y              ; update FCB
 
-    jmp select_active_drive
+    txa
+    jmp internal_LOGINDRIVE
 zendproc
```

I'll send a pull request after diner.

ivop, 2024-08-27:
Okay, that turned out to be the wrong fix, even though the problem description is right (i.e. all file related bdos_* functions call convert_user_fcb which then uses an active drive that is not logged in yet).

Another thing I noticed is this:

[image]

DIR B: silently changes the selected drive to B, but CCP does not update its prompt (global drive variable). Not even when I press enter. Only after CTRL-C does it notice it's no longer on A:. Same happens with TYPE B:somefile.txt

Perhaps CCP should call bdos_SELECTDSK with the drive variable as a parameter in the main loop? This automatically returns to what was the current drive before when calling an external application, too. @davidgiven ?

davidgiven, 2024-08-27:
`dir b:` shouldn't change the active drive.

From looking at the code, I think the BDOS isn't distinguishing between selecting the drive temporarily when working on an FCB on the non-current drive, and a call to LOGINDRIVE. I think BEGINSYS/ENDSYS need to store the old value of `current_drive` and restore it after the system call is done with, like the FCB drive is.

ivop, 2024-08-27:
In ccp.S, entry_DIR explicitly calls bdos_SELECTDISK. If I remove that, BDOS automatically uses the drive that's set in userfcb and then DIR B: does not change the active drive in CCP.  Transient commands, like stat, ls and dinfo, still change the active drive, but I suppose that's because they all call bdos_SELECTDISK, too, instead of relying on BDOS accessing the correct drive that is set in the passed fcb.

davidgiven, 2024-10-29:
Yup, I've just run into this myself on a new port. Investigating.

PR #200 (2024-10-30): Fix some BDOS and CCP issues.
- Ensure that the CCP and ls don't change drive when listing a directory.

This avoids issues where the current drive according to the BDOS doesn't match the current drive according to the CCP, which can make odd things happen.

Fixes: #151
Fixes: #113

- Fail with a hard error when trying to write to a full disk.

Previously it would just keep searching up off the end of the bitmap and trying to write to out-of-bound blocks.

Fixes: #170
closed 2024-10-30 by davidgiven