davidgiven/cpm65 #163: ATBASIC ERROR in string substrings

okwatts, 2024-09-05:
If you  try and extract a substring using A$(4,5) it appears that it is treated as a numeric and the left Bracket is used as exponentiation.
ie if B$="AB" and you print B$(1) you get 5.20200002e+53

ivop, 2024-10-03:
I can not reproduce this:

[image]

Could you show the exact steps you took and its output?

okwatts, 2024-10-04:
Here is a photo of my screen.

venomix666, 2024-10-05:
I can reproduce it on the BBC Micro (using b-em), see screenshot below.

[image]

venomix666, 2024-10-07:
I can also reproduce it on the Oric, VIC20 and PET (all emulated) - so somehow it seems to only  work on the Atari. Strange!

ivop, 2024-10-08:
I suppose some Atari dependency is still left in the source code (Altirra was originally written solely for the Atari), hence it working correctly on the Atari but not on other systems.

davidgiven, 2024-10-10:
I have vague memories that Atari basic uses hardware to generate random numbers, and that doesn't work on the CP/M-65 port, so it's plausible that there's some other Atari-specific code I've missed. But it's too long ago now and I'd have no idea what it is.

ivop, 2024-10-11:
Yes, it reads the Atari Pokey chip's RANDOM register, which is a 17-bit LFSR. I see you replaced that with `lda #0` in funRnd :smile: 

I could add sfc16 from [here](https://github.com/ivop/random6502). That would add better random numbers than the original on the Atari :wink:  A simple 16-bit LFSR could also be done, if adding 180 bytes would be too large.

This printing of a sub-string bug does not seem related though. I have no idea where to search for it.

eduardocasino, 2025-03-01:
I think I've found the problem.

Altirra Basic uses a variable, `expType`, to differentiate between numeric and string variables. From `atbasic.s`:

```
expType     dta     0       ;bit 7 = 0 for numeric, 1 for string
```

I've seen throughout the code that the developers try to save some bytes and cycles by assigning this variable with values intended for something else, like function tokens or address labels, which are known to have bit 7 set or cleared. For example, in this snippet from `functions.s`:

```
		;address += start
		ldy		#fr1+2
		sty		expType				;!! - set expression type to string!
		jsr		IntAddToFR0
```
Here, `Y` is an argument for `IntAddToFR0` and `fr1` is the address of the floating point register 1, which in the Atari 800 happens to be at ZP address `0xE0`, so it has bit 7 set.

The probem is that, in the CP/M port, those ZP addresses are not fixed and may be assigned to locations above or below `0x80`, depending on the memory map of the target system and it seems that, in many cases, it is below, so the expression type is being set to "integer" and printed as such.

I'm opening a pull request with this proposed fix:

```diff
--- a/third_party/altirrabasic/source/functions.s
+++ b/third_party/altirrabasic/source/functions.s
@@ -639,8 +639,9 @@ use_limit_as_end:
                ;##ASSERT dw(fr1+4) <= dw(fr0+4)

                ;address += start
-               ldy             #fr1+2
+               ldy             #$80
                sty             expType                         ;!! - set expression type to string!
+               ldy             #fr1+2
                jsr             IntAddToFR0

                ;capacity -= start
```

ivop, 2025-03-01:
Good catch! On the Atari port of CP/M-65, fr1 is still above 0x80, hence the problem does not manifest itself.

davidgiven, 2025-03-01:
Oof. Yikes. Thank you for finding this. That's the kind of optimisation which looks like a good idea until it isn't.