davidgiven/cpm65 #170: BBC crashes if all of bss is used

ivop, 2024-09-11:
I stumbled upon this error with a new program I'm writing, and can reproduce it with copy:

[image]

All other ports seem to work correctly when all of bss is used and the program calls cpm_warmboot() after that.

davidgiven, 2024-09-12:
This sounds like the DPH is misconfigured and set up for a bigger disk than MOS actually has. MOS then tries to expand the disk file, fails, and throws an error. MOS errors all use quasi-exception handling via brk and there is no brk handler to catch them, which is why it prints the message and crashes. Both of these things need fixing.

davidgiven, 2024-10-29:
I've encountered a similar problem elsewhere; turns out the BDOS doesn't check for disk full conditions when allocating new blocks and will just run off the end of the bitmap. Oops!

PR #200 (2024-10-30): Fix some BDOS and CCP issues.
- Ensure that the CCP and ls don't change drive when listing a directory.

This avoids issues where the current drive according to the BDOS doesn't match the current drive according to the CCP, which can make odd things happen.

Fixes: #151
Fixes: #113

- Fail with a hard error when trying to write to a full disk.

Previously it would just keep searching up off the end of the bitmap and trying to write to out-of-bound blocks.

Fixes: #170
closed 2024-10-30 by davidgiven