davidgiven/cpm65 #171: Apple 2 port stopped working

ivop, 2024-09-11:
The Apple 2 port stopped working somewhere in the past. I don't know when it last worked, but at the moment it hangs after the CP/M-65 banner.

[image]

andreasbaumann, 2024-09-12:
You might hit a max-size limit (I did in the past). IIRC the monitor_command in ccp.S makes things just a little bit too big to be loaded.

```
         ; Monitor command?
-        cmp #'@'
-        beq monitor_command
+        ;~ cmp #'@'
+        ;~ beq monitor_command
```

and then commenting out the part after `zproc monitor_command` worked for me
as a temporary workaround.

ivop, 2024-09-12:
I can confirm that the apple port indeed starts working again if the monitor command and associated procedures are commented out. And then I found out that the Apple ][ port has no screen driver, so it's not needed for my tests :)

davidgiven, 2024-09-12:
The CCP, BDOS and BIOS all preferentially load into high RAM, and the Apple II only has 8kB of it. I guess that this change made it run out. It's _supposed_ to fall back to using low RAM instead but I forget whether I actually implemented that bit yet.

In the meantime I'll add a tiny variant of the CCP, without the monitor, and make the Apple II use it. I'll leave the bug open for looking at the low RAM stuff.

Note: I don't currently have a working Apple emulator --- mame's Apple II support seems to be non-functional. So this has had no testing.

andreasbaumann, 2024-09-13:
@ivop: Actually, I was working on an Apple screen driver and 40/80-columns mode handling in a branch https://github.com/andreasbaumann/cpm65/tree/apple2_plus. Should take it up again and sync it with the main branch probably.

ivop, 2024-09-13:
Yes, a screen driver for the Apple port would be very nice!

As for emulation on Linux, I run linapple which seems to work quite well.

closed 2024-09-16 by davidgiven

andreasbaumann, 2024-09-17:
My memory is fading apparently. :-)

I merely adapted the existing 80-column drivers (TTY and SCREEN) to work also
on older machines with 40-columns. Sorry about that mistake. I didn't want to take
credit for anything here.

davidgiven, 2024-09-17:
Very unlikely, I'm afraid --- sorry, I didn't register that you were working on this. Feel free to send me a PR anyway and I'll manually merge 40-column support.

One annoying thing is that the standard Apple character set only has partial support for inverse text so that without the IIe's extended character set, inverse text will look very weird (capitals only!).

andreasbaumann, 2024-09-17:
No worries, I didn't touch the branch since March, and I almost forgot, what I did in
there (40/80 column mode, some preliminary screen driver and a tetris of course to
test the screen driver). :-)

I'll merge and make more small pull request for the individual things.

I remember one especially tricky thing was to reuse code from the tty driver
also in the screen driver..

The inverse mode I opted to print only a space in inverse when blinking the
cursor (which is also not nice). And to do all capital seems to be the only option
in 40-columns mode too (IIRC I didn't implement it at all making the qe status bar
look  weird).

To complicate things further, a Videx 80-column card for instance can do proper
inverse mode for all characters..

andreasbaumann, 2024-09-17:
Created a pull request https://github.com/davidgiven/cpm65/pull/177

PR #176 (2024-09-16): Add an Apple SCREEN driver. Fix CCP loading with no high memory.
Fixes: #171