davidgiven/cpm65 #187: just questions

norayr, 2024-10-07:
* which filesystem is used by cp/m? it has own filesystem, right? does it have directories? if it uses own file system, then it is not able to read disks formatted with native oric, atari, etc tools?
* programs written for other operating systems won't run on cp/m, but programs written for cp/m for 6502 will run the same on all supported platforms? then probably that means no high resolution modes but just alphanumeric screen? then what if different machines have different number of charachers per line or different number of lines?
* does cp/m load its object files or those object files should be linked in to some other format to be run?

davidgiven, 2024-10-08:
Re filesystem: yes, it uses it's own (the standard CP/M file system). It doesn't have directories, and no, it can't access host files. Typically the host filesystem isn't running at all. The exceptions are the Neo6502 (where the CP/M filesystem is emulated on top of the existing MS-DOS filesystem), the BBC Master and Tube platforms (where the CP/M filesystem is stored in a file accessed via the host filesystem) and the Commodore platforms (where the host filesystem is running in an entirely different computer).

Re screen: yes, text mode only. There's a screen driver which allows you to query things like the screen size.

Re object files: not sure what you're asking here. CP/M-65 binaries are special relocatable files rather than memory images. You do need the right tooling to create these.

norayr, 2024-10-09:
thank you! i think it answers my questions. thank you for the great work.

closed 2024-10-09 by norayr