davidgiven/cpm65 #188: Some doc updates (not an issue, but useful info for building)

DevTheo, 2024-10-08:
I just fired up WSL with Debian to try to build this project on Windows (well, really Debian under Windows)

There were some (likely pretty basic) packages that were needed for me to be able to build (I went back through my shell history to determine what they were and wrote them all down).
First off all the items already listed in the readme: 
`cc1541 cpmtools libfmt-dev python3 fp-compiler srecord`

Additionally I needed the following items
xz-utils - for extracting some of the tar files you will run into
make
pkg-config
libreadline-dev
gcc
build-essential
compiler)
moreutils - for chronic
gawk
zip

Additionally I had to manually install FreePascal (via fpc-3.2.2.x86_64-linux.tar from their SF site), `sudo apt-get fp-compiler` didn't get it (assuming that's the FreePascal). I also installed wget (but that was so I could paste in a uri to a file and get it downloaded to my linux subsystem --without resorting to copying from the Windows drive)
 
So my hopes are to get a C compiler running within CPM65 (and possibly also get C128 to be able to run CPM65). To start with I am going to try to port over the assembly code from the _other_ CPM65 project (someone already did this, but I don't think it is as good as this one.. I will make sure that the license is compatible before even thinking of pushing a pull-request).

davidgiven, 2024-10-08:
Cheers --- I should update the dependencies. The source of truth is actually the github CI scripts in .github/workflows.

Re a C compiler: I actually got a little way into porting Alan Cox's Fuzix Compiler Kit. Look in the `cc` branch. The compiler binaries build on the host, but don't cross-compile with llvm-mos yet because llvm-mos's standard library doesn't contain stuff like fopen.

Re porting the assembler: be aware that CP/M-65's relocatable binaries requires a rather different approach to assembling than the traditional CP/M memory image assembler. You may not be able to port a third party one.

DevTheo, 2024-10-08:
Thanks for the quick response. The big thing is I have recently gotten interested in CPM using a C128 emulator (long story that I won't go into). Right now I am willing to see what I can do to try to help the project along. I'm definitely a retro-geek (which means both gaming and these programming old systems). I primarily use an emulator to play around with this stuff (I'm not sure too many people are using actual hardware). BUT, I want to see what I can get running (so the steps to try a different set of source is really me familiarizing myself with programming here.. and remembering 6502 assembly)

DevTheo, 2024-10-08:
Actually, let's close this.. I never meant for it to be "open" just for it to available for anyone else trying to get this to build.

closed 2024-10-08 by DevTheo