davidgiven/cpm65 #222: Oric multiple drive support

mihai-dragan, 2025-02-21:
It says in the README "Only the first drive is supported. This would be easy to extend if anyone's interested."
I found the "bios_SELDSK" procedure in oric.s but I don't know much assembler, so I tried to see how this is implemented for other targets and I see most of them only support one drive too. Where should I look for a good example of multiple drives support? And can you give me any pointers on what must be done?

davidgiven, 2025-05-20:
Hello --- sorry for the long delay, been absorbed in other projects... this was slightly waiting for another piece of work which I've just done.

Essentially, what needs to happen is:

- define two DPHs (one for each drive) referencing the same DPB.
- change bios_SELDSK to return the appropriate DPH based on which drive the user asks for.
- change the actual read/write implementation to pay attention to the drive number and set the relevant bit in the drive controller to operate on the right drive.

The latter's a bit complicated due to the Oric BIOS using a buffer for the current sector; this is needed because CP/M deals with 128-byte records and the disk format used 512-byte sectors. So, when changing drives you'd need to flush and invalidate the buffer. Flushing the buffer can be done by calling `flush_buffer` and invalidating it can be done by setting the currently buffered sector or track to something impossible --- look in `_start` for where it's setting these to 0xff. (Invalidation is necessary to force the new sector to be read off the other drive, rather than using the buffered value from the old drive, which of course is wrong.)

Let me know if you're still interested in doing this and I'll see what I can do to help!

iss000, 2025-05-21:
+1 for the interest in multi drive support! I'm ready to help and test.