davidgiven/cpm65 #226: Pascal-M readln from console does not work

mihai-dragan, 2025-03-07:
Program to reproduce:
program Test;
var
    s : packed array [1..100] of char;
begin
    readln(s);
    writeln(s);
end.

With the latest version of the interpreter (with LF for endline) looks like EOLN is never detected, but even with the previous version where the readln would return when enter is pressed, the string would not be filled.
We know this works when reading from files because the compiler uses readln for that and it works.

eduardocasino, 2025-03-09:
The fix for the `CR`/`LF` thing is pretty straightforward:

```diff
--- a/third_party/pascal-m/pascalmint2k1.S
+++ b/third_party/pascal-m/pascalmint2k1.S
@@ -2317,7 +2317,10 @@ rdeof:  lda #$1a
         rts

 ttyin:  jsr bdos_CONIN
-        ldy ytmp
+        cmp #CR
+        bne ttyin1
+        lda #LF
+ttyin1: ldy ytmp
         rts
 ;
 ; Write one char to TTY
```

However, it seems that `readln()` into strings is not working. You can see that when there is an error in a pascal program, when a bunch of empty lines is displayed instead of the error details.

I'll have a look at it and try to make a pull request with both fixes, unless somebody else takes the task ;)

mihai-dragan, 2025-03-11:
The maximum limit of 256 characters is not a big problem, in my opinion. The fact that you don't know where the user input ends and the "filler" begins, though... that is an issue. Can't we use a terminator like chr(0)? Also, filling with space seems an unnecessary overhead, especially if there is a clear terminator char.

eduardocasino, 2025-03-11:
The real solution would be implementing the `string` type, anything else has its advantages and disadvantages. See the discussión in pull request #229.

The standard Pascal just stops filling when receives an EOL and the rest of the array is left unmodified. I added the filling just for convenience, but it can be removed. 

It is also standard Pascal behaviour to fill with spaces the remaining of the array if you assign to it a shorter screen.

From the Pascal-M changelog:

```
    Additions for version 1.3 :

    - standard functions SUCC and PRED added
    - array of character constants and variables of any length
      may be assigned to variables of type(packed)array of char
      if length is less or more than nr of characters in array
      The result is padded with blanks or truncated
    - character may be assigned to array of char,
      result padded with blanks
    - bug in body(complex structures formal type corrected
```

The overhead is just some processor cycles, as the space is already occupied. The `packed` attribute is simply ignored in Pascal-M.

closed 2025-03-20 by mihai-dragan